TitusTech/tia-portal-automation-tool #62: Site_Builder - Add new page "Devices" for page navigation

wine-ff2, 2025-04-25:
**<h2>USER STORY / DESC </h2>**
I want to add a new view page for "Devices" which will contain the necessary items and fields of the devices table.

[image]

**<h2>MOCKUP</h2>**

**<h2>ACCEPTANCE CRITERIA</h2>**

**<h2>PRIORITY/SEVERITY</h2>**

**<h2>DEV-TEST ESTIMATE</h2>**

wine-ff2, 2025-04-28:
[image]

allana-titusgt, 2025-05-19:
https://github.com/TitusTech/tia-portal-automation-tool/issues/61#issuecomment-2890068499

allana-titusgt, 2025-05-20:
The new view page for the Device Category homepage is displaying correctly, and all functionalities are operating as intended.

[image]