TitusTech/tia-portal-automation-tool #69: Site Builder - Redesign the automation tool JSON Configuration

mdjmroque, 2025-05-13:
**<h2>USER STORY / DESC </h2>**

**<h2>MOCKUP</h2>**

**<h2>ACCEPTANCE CRITERIA</h2>**

**<h2>PRIORITY/SEVERITY</h2>**

**<h2>DEV-TEST ESTIMATE</h2>**

wine-ff2, 2025-05-13:
```json
      "id": 1,
      "p_name": "PLC_1",
      "p_typeIdentifier": "OrderNumber:6ES7 512-1DK01-0AB0/V2.6",
      "p_deviceName": "PLC_1",
      "network_interface": {
        "Address": "192.168.88.210",
        "RouterAddress": "192.168.88.1",
        "subnet_name" : "PN/IE_1",
        "io_controller" : "PNIO"
      },
      "slots_required": 0
```

wine-ff2, 2025-05-16:
```
Unhandled Exception: Siemens.Engineering.EngineeringSecurityException: Security error.
Can't access this process to acquire needed information.
Security error.
Owner 'username' of this process is not member of the windows group 'Siemens TIA Openness'.
Please contact your administrator.
```

wine-ff2, 2025-05-16:
[template.json](https://github.com/user-attachments/files/20240581/template.json)

[SMC_Cebu.json](https://github.com/user-attachments/files/20240587/SMC_Cebu.json)

wine-ff2, 2025-05-19:
NEW TASK:
- Site_Builder add new category:
	- Local modules
	- in Devices tags, break up network_interface into different columns
		- routeraddress is optional, use button
		- check on subnet mask
- prio internal tooling (tia portal)
	- overwrite button in Project tab
	- remove it in json

wine-ff2, 2025-05-23:
[image]